meshtastic/Meshtastic-Android #1097: Enhancement : If unknown node - show HEX nodeID

Dorn8010, 2024-06-14:
Instead of showing Unknown Username, show node ID instead.

This would open the way to the next step to write to a certain nodeID with the app or CLI command.
This would allow to address e.g. a known node with a new device.

djripcord, 2024-06-16:
I agree this would be a great way of showing it. Helps especially when chatting with multiple nodes that you haven't received telemetry for yet. Instead of every node being anonymous, you at least get some information that helps determine who you're talking with.

roha-github, 2024-06-20:
You could also enter the long name manually for Unknown Nodes.

Example: the NodeID is 1f5a and I find out in the chat that this is Michael, then I could enter the long name "Michael" in the app. As soon as a node info is received, the long name is corrected.

PR #1144 (2024-07-28): Expanding node info cards
This is a rework of the UI for NodeInfo cards to be expandable. Tapping the card now expands the view to show less pertinent details as discussed in discord, https://github.com/meshtastic/Meshtastic-Android/pull/1111#issuecomment-2197270357, and #1138. A global show details toggle is available in the filter menu.
As part of this -  the views were reworked from a Constraint view into Rows for future readability and maintainability.

Hides the coordinates by default as mentioned in @garthvh's comment here https://github.com/meshtastic/Meshtastic-Android/discussions/1138 [... cut]
closed 2024-07-28 by andrekir