meshtastic/Meshtastic-Android #1100: [Feature Request]: Android APP sorting and filtering

TheDomanNow, 2024-06-17:
### Platform

ESP32

### Description

Hi,

after app update 2.3.11, there is some sorting added, but... I think it's still missing most important thing:
1. Show only LoRa radio connections, _without_MQTT_ (if i have LoRa+MQTT enabled). Why? When I have LoRa and MQTT enabled (say 50 nodes via MQTT), and I'm doing some tests on local radios, I have to select "Ignore MQTT" in radio, then reset the NodeDB, and then wait for ages... all of this can be avoided by just showing LoRa radio nodes/connections.
2. Modifier/filter to always show LoRa radio nodes first, and then, after all LoRa radio nodes have been listed, and only then, list MQTT. This would be extreme quality od life improvement (to me at least).

Please :-)

BR, Doman.

Talie5in, 2024-06-18:
I would like to add to this request, when sorting by Hops, devices that have been unheard for a few hours (technically unreachable/offline) Sit above Hop Count 1 devices, these should be shuffled to the bottom.
Suggestion is anything not heard for an hour or more.

andrekir, 2024-06-30:
Duplicate of #1071

closed 2024-06-30 by andrekir