meshtastic/Meshtastic-Android #1101: Facing build issue with Meshtastic-Android-2.3.11 on Android Studio Koala | 2024.1.1

gawadepd07, 2024-06-18:
e: [ksp] [MissingType]: Element 'com.geeksville.mesh.database.Converters' references a type that is not present
e: [ksp] D:/Meshtastic/Meshtastic-Android/app/src/main/java/com/geeksville/mesh/NodeInfo.kt:26: [MissingType]: Element 'com.geeksville.mesh.MeshUser' references a type that is not present
e: [ksp] D:/Meshtastic/Meshtastic-Android/app/src/main/java/com/geeksville/mesh/NodeInfo.kt:67: [MissingType]: Element 'com.geeksville.mesh.Position' references a type that is not present
e: [ksp] D:/Meshtastic/Meshtastic-Android/app/src/main/java/com/geeksville/mesh/NodeInfo.kt:128: [MissingType]: Element 'com.geeksville.mesh.DeviceMetrics' references a type that is not present
e: [ksp] D:/Meshtastic/Meshtastic-Android/app/src/main/java/com/geeksville/mesh/NodeInfo.kt:153: [MissingType]: Element 'com.geeksville.mesh.EnvironmentMetrics' references a type that is not present
e: [ksp] D:/Meshtastic/Meshtastic-Android/app/src/main/java/com/geeksville/mesh/database/entity/MeshLog.kt:13: [MissingType]: Element 'com.geeksville.mesh.database.entity.MeshLog' references a type that is not present
e: [ksp] D:/Meshtastic/Meshtastic-Android/app/src/main/java/com/geeksville/mesh/database/dao/NodeInfoDao.kt:15: [MissingType]: Element 'com.geeksville.mesh.database.dao.NodeInfoDao' references a type that is not present
e: [ksp] androidx.room.RoomKspProcessor was unable to process 'com.geeksville.mesh.database.MeshtasticDatabase' because not all of its dependencies could be resolved. Check for compilation errors or a circular dependency with generated code.
e: [ksp] InjectProcessingStep was unable to process 'ChannelSetRepository(androidx.datastore.core.DataStore<error.NonExistentClass>)' because 'error.NonExistentClass' could not be resolved.

Dependency trace:
    => element (CLASS): com.geeksville.mesh.repository.datastore.ChannelSetRepository
    => element (CONSTRUCTOR): ChannelSetRepository(androidx.datastore.core.DataStore<error.NonExistentClass>)
    => type (EXECUTABLE constructor): (androidx.datastore.core.DataStore<error.NonExistentClass>)void
    => type (DECLARED parameter type): androidx.datastore.core.DataStore<error.NonExistentClass>
    => type (ERROR type argument): error.NonExistentClass

If type 'error.NonExistentClass' is a generated type, check above for compilation errors that may have prevented the type from being generated. Otherwise, ensure that type 'error.NonExistentClass' is on your classpath.
e: [ksp] InjectProcessingStep was unable to process 'LocalConfigRepository(androidx.datastore.core.DataStore<error.NonExistentClass>)' because 'error.NonExistentClass' could not be resolved.

Dependency trace:
    => element (CLASS): com.geeksville.mesh.repository.datastore.LocalConfigRepository
    => element (CONSTRUCTOR): LocalConfigRepository(androidx.datastore.core.DataStore<error.NonExistentClass>)
    => type (EXECUTABLE constructor): (androidx.datastore.core.DataStore<error.NonExistentClass>)void
    => type (DECLARED parameter type): androidx.datastore.core.DataStore<error.NonExistentClass>
    => type (ERROR type argument): error.NonExistentClass

If type 'error.NonExistentClass' is a generated type, check above for compilation errors that may have prevented the type from being generated. Otherwise, ensure that type 'error.NonExistentClass' is on your classpath.
e: [ksp] InjectProcessingStep was unable to process 'ModuleConfigRepository(androidx.datastore.core.DataStore<error.NonExistentClass>)' because 'error.NonExistentClass' could not be resolved.

Dependency trace:
    => element (CLASS): com.geeksville.mesh.repository.datastore.ModuleConfigRepository
    => element (CONSTRUCTOR): ModuleConfigRepository(androidx.datastore.core.DataStore<error.NonExistentClass>)
    => type (EXECUTABLE constructor): (androidx.datastore.core.DataStore<error.NonExistentClass>)void
    => type (DECLARED parameter type): androidx.datastore.core.DataStore<error.NonExistentClass>
    => type (ERROR type argument): error.NonExistentClass

If type 'error.NonExistentClass' is a generated type, check above for compilation errors that may have prevented the type from being generated. Otherwise, ensure that type 'error.NonExistentClass' is on your classpath.
e: [ksp] AssistedInjectProcessingStep was unable to process 'MockInterface(com.geeksville.mesh.repository.radio.RadioInterfaceService,java.lang.String)' because 'error.NonExistentClass' could not be resolved.

Dependency trace:
    => element (CLASS): com.geeksville.mesh.repository.radio.MockInterface
    => element (METHOD): handleAdminPacket(error.NonExistentClass,error.NonExistentClass)
    => element (PARAMETER): pr
    => type (ERROR parameter type): error.NonExistentClass

If type 'error.NonExistentClass' is a generated type, check above for compilation errors that may have prevented the type from being generated. Otherwise, ensure that type 'error.NonExistentClass' is on your classpath.
e: [ksp] AssistedFactoryProcessingStep was unable to process 'com.geeksville.mesh.repository.radio.MockInterfaceFactory' because 'error.NonExistentClass' could not be resolved.

Dependency trace:
    => element (CLASS): com.geeksville.mesh.repository.radio.MockInterface
    => element (METHOD): handleAdminPacket(error.NonExistentClass,error.NonExistentClass)
    => element (PARAMETER): pr
    => type (ERROR parameter type): error.NonExistentClass

If type 'error.NonExistentClass' is a generated type, check above for compilation errors that may have prevented the type from being generated. Otherwise, ensure that type 'error.NonExistentClass' is on your classpath.
e: [ksp] AssistedProcessingStep was unable to process 'address' because 'error.NonExistentClass' could not be resolved.

Dependency trace:
    => element (CLASS): com.geeksville.mesh.repository.radio.MockInterface
    => element (METHOD): handleAdminPacket(error.NonExistentClass,error.NonExistentClass)
    => element (PARAMETER): pr
    => type (ERROR parameter type): error.NonExistentClass

If type 'error.NonExistentClass' is a generated type, check above for compilation errors that may have prevented the type from being generated. Otherwise, ensure that type 'error.NonExistentClass' is on your classpath.
e: [ksp] ModuleProcessingStep was unable to process 'com.geeksville.mesh.repository.datastore.DataStoreModule' because 'error.NonExistentClass' could not be resolved.

Dependency trace:
    => element (OBJECT): com.geeksville.mesh.repository.datastore.DataStoreModule
    => element (METHOD): provideLocalConfigDataStore(android.content.Context)
[... 24 lines cut ...]
Dependency trace:
    => element (OBJECT): com.geeksville.mesh.repository.datastore.DataStoreModule
    => element (METHOD): provideLocalConfigDataStore(android.content.Context)
    => type (DECLARED return type): androidx.datastore.core.DataStore<error.NonExistentClass>
    => type (ERROR type argument): error.NonExistentClass

If type 'error.NonExistentClass' is a generated type, check above for compilation errors that may have prevented the type from being generated. Otherwise, ensure that type 'error.NonExistentClass' is on your classpath.
e: Error occurred in KSP, check log for detail

> Task :app:kspFdroidDebugKotlin FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:kspFdroidDebugKotlin'.
> A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction
   > Compilation error. See log for more details

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

BUILD FAILED in 1m 13s

gawadepd07, 2024-06-19:
After fetching and updating the repo issue has been resolved. 

Thank you...

closed 2024-06-19 by gawadepd07

kirinn600, 2025-09-26:
HI. Can you explain a bit more please? exactly, how this was fixed?
thank you

jamesarich, 2025-09-26:
> HI. Can you explain a bit more please? exactly, how this was fixed? thank you

this issue is when the repository is not able to find the Protobufs, which usually means the repo was probably not recursively checked out with the `proto` submodule as the docs recommend.

`git clone https://github.com/meshtastic/Meshtastic-Android.git --recursive && cd Meshtastic-Android/`

https://meshtastic.org/docs/development/android/