meshtastic/Meshtastic-Android #1115: Cache node names on app

mudcamper, 2024-06-29:
The latest 100 node names live in the node dB on the node, but it would be useful if ALL nodes heard (or the last 1000 or something) would be saved (at least short and long name) on the app. That way when "new" nodes show up, the name can be quickly found from the node name dB cache on the app rather than showing as "unknown" until the name resolves.

Where I live (north SF Bay Area) there are so many nodes that all my local nodes drop off the list because of the 200+ nodes (which are too far away to be useful) are so active they push all my local nodes off the list of 100.

andrekir, 2024-06-30:
apps/clients are just a remote view into the connected node which is the source of truth for the node database.

closed 2024-06-30 by andrekir

mudcamper, 2024-07-03:
andrekir Yes the info is stored on the nodes. But where I live, hundreds of nodes constantly flush through the node dBs, removing all my local nodes from the node dBs. Then when a message comes from one of these local nodes, it is "unknown". It may eventually resolve, but it would be useful if the app, not the node, stored all the names it has ever seen, so that when an unknown message comes, it looks up the name using the hardware ID and it's larger name dB. Maybe it would be out of date, but it's better than "unknown" or "????" and it would get updated if/when the node dB updates. 

This is a real problem in areas where hundreds of nodes constantly flush out your local nodes. All the nodes I ever want to communicate with are always gone from the node dB, which is filled with useless far away active nodes. 

Why did you close this?