meshtastic/Meshtastic-Android #1125: Multiple DM Window/Entries

fti7, 2024-07-03:
Testing with https://github.com/TheCommsChannel/TC2-BBS-mesh
DM to "ftb1" with "hey"
Instead of the Response in the same Window/Entry, i got a new one.

Seems like the "to" ID is corrupt. Afaik i went to the DM from the Nodes Tab. I cant really reproduce it. But it happens from time to time.
App Cache/Storage Client didnt fixed it

[image]
[image]

andrekir, 2024-07-06:
different channels. received message came in your primary channel, DM was sent over your channel 1.

closed 2024-07-06 by andrekir

fti7, 2024-07-06:
> different channels. received message came in your primary channel, DM was sent over your channel 1.

thx, ah ok.... that makes sense. So its an TC2-BBS Issue.
Is there any way i can influence with which channel it sends? or is it random? that device and i have both configured (channel 0 +1). Maybe it should also show in the DM Window which channel it uses

CommanderCrash, 2024-12-24:
This bug still happens and if you have LongFast it will always default it to that instead.