meshtastic/Meshtastic-Android #1136: [Bug]: Times in node list do not update

RichieFrame, 2024-07-14:
When viewing the node list, the last seen time appears to be essentially static unless a remote node is seen, then the time will update to 'now'. None of the other nodes will count their time up on a refresh interval. A node may say last seen 'now' but the actual last transmission may have been 20 mins ago, or even hours. Terminating the app and reopening is the only way to get the times to update.

b8b8, 2024-07-14:
Switching between map and nodelist tab or messages and nodelist tab seems to update this time for me without having to restart the app.

RichieFrame, 2024-07-16:
If that was the case I might not have noticed it, but alas that is not the behavior on my phone... however based on your comment I started clicking everything, switching the color theme of the app will cause the update to occur, so at least there is a workaround

RichieFrame, 2024-07-30:
I just installed 2.4.0 on a different phone and can confirm the issue is still present

b8b8, 2024-08-03:
it is still an issue for me as well.

neoterix, 2024-08-03:
+1 adding another data point for observing this issue (Android version 2.3.15, Pixel 8 Pro). Force close works, switching tabs does not.

Ultrahighfrequency, 2024-08-03:
I get this issue also

todd-herbert, 2024-08-17:
Hey I've got 2.4.2 alpha from play store and I'm still seeing this. Is there info I can gather to help out with this or anything? Bit out of depth when it comes to app dev stuff..

Talie5in, 2024-08-18:
@todd-herbert is frequent since app 2.4.2, but  where I saw some stickiness seems to be local cache of nodedb, did a full app stop, clear data, start app and connect likes its the first time to a device.

It cleared up some stuck time updates..

Of note, the ones that were stuck showed correct last heard time.on the map for me.
Switching between map and node list didn't unstuck these ones.

andrekir, 2024-08-18:
the root cause for this issue (and what was addressed in #1178) is that the UI only updates when a value changes. so when the `lastHeard` value for a node is static, we need to kick it to update with the calculated "time ago" value (time now - lastHeard).

a good test to confirm if this is indeed not solved is to change screen orientation between portrait/landscape. if the time updates, there is still an issue. if not, it is something else. I'm still unable to reproduce this.

todd-herbert, 2024-08-18:
I should probably have been specific about what it was I've noticed: resuming the app (after a period of inactivity) shows stale times. Toggling the device theme in-app has been my workaround so far.

> something to note: currently the time takes 1 minute to update after waking up the phone from stand-by.

Does that apply in this case of resuming the app? Just so I know if it's expected behavior.

___

I something to admit: I just checked my phone's battery settings, and apparently Meshtastic was no longer added to the list of "Never auto sleeping apps". I must have unintentionally removed it a few weeks back when testing different versions.

I've returned it to the list of "Never auto sleeping apps", and will keep an eye on it over the next day or so. I could easily imagine that being the source of issues on my end.. 

I'll test out changing the screen orientation too and get back to you. Thanks for your time!

andrekir, 2024-08-18:
> > something to note: currently the time takes 1 minute to update after waking up the phone from stand-by.
> 
> Does that apply in this case of resuming the app? Just so I know if it's expected behavior.

yes. I just pushed some code to address this though, as updates should be immediate once the app resumes.

> 
> (...) apparently Meshtastic was no longer added to the list of "Never auto sleeping apps".

even if the app was killed, once the node screen loads, all data should be up-to-date.

today's changes are available in a debug APK here:
https://github.com/meshtastic/Meshtastic-Android/actions/runs/10441480822/artifacts/1824752386

if someone is able to install and test, it would be super helpful.

todd-herbert, 2024-08-18:
> if someone is able to install and test, it would be super helpful.

Running it here now, and it seems like the times are updating straight away when resuming the app. I think you got it!

andrekir, 2024-09-01:
this should be fixed in recent releases. closing.

PR #1178 (2024-08-04): Fix: Periodically update last heard time
This introduces a periodic update to the last heard time displayed in the UI. The time is now refreshed every 1 minute
 to ensure accurate and up-to-date information.

fixes #1136
closed 2024-09-01 by andrekir