meshtastic/Meshtastic-Android #1140: [Bug] Times in map view doesn't update

kittenarmy, 2024-07-14:
Issue: One nearby node 1 hop away (via my other node) is showing 48d next to their name, however in the node list it appears 1min and I can traceroute it. I tried closing the Android app manually. I also tried clearing nodeDB on both my devices.

I don't know why this nearby node won't update because my other node is changing by minutes each time I reopen the app (now, 5 mins) . My nodes are running latest heltec-v3 2.3.13.

prokrypt, 2024-07-14:
Other node does not have a valid time set.

andrekir, 2024-07-21:
map shows the **timestamp** of the position packet. same as messages show a timestamp. nothing to do with nodes "last heard".

kittenarmy, 2024-07-21:
> map shows the **timestamp** of the position packet. same as messages show a timestamp. nothing to do with nodes "last heard".

Thanks for clarifying. So I am struggling to see the utility of it showing such an old timestamp for nodes (seems to be max 99d) on the map when they also appear recently in my node list with coordinates. 

Other than ignoring these nodes altogether, how to get them to refresh or synchronise with their apparent location in node list? Do I have to manually request position for each stuck node?

b8b8, 2024-08-18:
> > map shows the **timestamp** of the position packet. same as messages show a timestamp. nothing to do with nodes "last heard".
> 
> Thanks for clarifying. So I am struggling to see the utility of it showing such an old timestamp for nodes (some exceed 90 days) on the map when they also appear recently in my node list with coordinates.
> 
> Other than ignoring these nodes altogether, how to get them to refresh or synchronise with their apparent location in node list? Do I have to manually request position for each stuck node?

They would have to send another position packet, or you would have to request their location manually.

kittenarmy, 2024-09-01:
> They would have to send another position packet, or you would have to request their location manually.

Ah that's a bit annoying. I wonder if there is a way to purge these nodes from the database as somehow they keep getting added back even after I reset my own node.db