meshtastic/Meshtastic-Android #1153: FEATURE REQUEST | Privacy warning on first message to public mesh

wayneoutthere, 2024-07-22:
First, amazing project - thank you so much.

Second: disclaimer: I've not actually used this app at all, however, while asking in a group some privacy questions it was said that 'any message sent on public mesh, even by DM is not encrypted and can be ready by anyone".  I asked if the user was told about this in GUI and they said 'no', so I wanted to post this quick idea for all the GUI for meshtastic (iOS, web, etc).  Not sure where to post it so posting here.

When the user, for the first time only, sends the first DM or public mesh message, greet them with a quick dialogue window alert that looks something like this:

**"Anyone can read the contents of this message (not encrypted).  Proceed?"  Y/N**

Then add the 'don't warn me again about this' option to the alert window so the nag doesn't continue if they don't want.

That way no one will accidentally send information they don't want (ie. Journalist in a hostile regime).

Hope this idea helps.

Thanks so much!

madeofstown, 2024-08-20:
The warning should happen before you set the region. Otherwise, node information and GPS location data will still be sent. Maybe just a warning on the Play Store would be good enough but I think Meshtastic is kind of a niche application where people mostly understand what they are doing when they buy a node and run the software.

wayneoutthere, 2024-08-20:
Then the first bug we have to fix is that: that it's niche.  If we could get grandma and grandpa messaging us with this, we'd have a thing ;)

jp-bennett, 2024-08-22:
Rather than add the message, we added PKI for DMs in version 2.5. position settings will now default to a fuzzy position, too.

 It might be worth making it a little more obvious that the default channel has a default key that everyone knows.

madeofstown, 2024-08-22:
> Then the first bug we have to fix is that: that it's niche. If we could get grandma and grandpa messaging us with this, we'd have a thing ;) Anything would help, though, for now.

I disagree, when the Internet became a commodity that's when it stopped being a fun and open community. Niche is what makes community meaningful.

wayneoutthere, 2024-08-22:
This will veer off topic a bit, but probably worthwhile for a post or two: A very fair veiw point you have.  To add a bit of perspective to where I am entering this is as follows: I had (barely) started learning Python a few years ago.  When I saw Meshtastic it reminded me of of the vision I had for mesh networks.  The exact project that inspired me a few years back was [serval](https://servalproject.com/).  It kind of 'died on the vine' but I never forgot it.  Meshtastic just re-inspired me and I started dusting off my python (as a starting point even though it seems it won't be super useful?) and the main reason was this: there is no app for [ubuntu touch](https://ubuntu-touch.io/) and only for the (very) compromised systems of Android and iOS.  So, to me, Meshtastic, other than the browser based version, is not starting out 'free' which is concerning.  Anyways, so I'm starting this more from a freedom perspective where grandma and grandpa are just victims of the commodity (which you fairly called it) and deserve a way to communicate with out it, freely.  hope this helps a bit to show what I'm aiming at here.