meshtastic/Meshtastic-Android #1169: Mestastic App Bluetooth Broken on Pixel Phones

andymccall, 2024-07-31:
Hi,

I've got a 2 x Lilygo T Decks and 2 x Heltec V3 (3.1)'s. I have flashed them with 2.4.0.46d7b82 Beta firmware.

Connecting to them via bluetooth using Pixel phones fails. I have tested this with a Pixel 2 XL and a Pixel 6 Pro

The device appears to pair, prompting for the PIN and then and shows as paired in the phone settings, but the MAC address of the device shows in the Meshtastic app and it is dark grey.  You cannot select the device.

I have checked the devices themselves are working by connecting to them via Bluetooth on a Motorola phone, which worked.

I have uninstalled the Meshtastic app, cleared the data, reset the phone turned bluetooth on an off etc. however there is no way to connect using these phones, so I think it is related to Google's Android OS.

gjelsoe, 2024-07-31:
I had the same issue but somehow I got it going again.
I'm on a Pixel 7 Pro, last update was on the 5. July 2024

andymccall, 2024-07-31:
I've managed to work it out.

It's the Precise Location. If you turn this off then it will list the devices. Once you've connected to the device and done a few things, you can turn Precise Location back on and it works going forward.

It looks like there's something related to Precision Location on Pixel phones that's causing it to hang on adding devices over Bluetooth.

You can add them via IP fine, so it's specifically Bluetooth and Precise Location on Pixel phones.

gjelsoe, 2024-07-31:
Something has changed.
I had 4 devices already connected to my Pixel and when I tested some
new code and had issues with connecting a new devices didn't have any problems with the previous version.

garthvh, 2024-07-31:
In certain versions of android BLE can be used to triangulate user location so it requires location permissions.

andymccall, 2024-08-01:
> In certain versions of android BLE can be used to triangulate user location so it requires location permissions.

The manifest of the app details this:

`    
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN"
        android:usesPermissionFlags="neverForLocation"
        tools:targetApi="s" />`
        
Could this be denying BLE permission to triangulate the user location?

andymccall, 2024-08-01:
> FWIW, running the latest GrapheneOS on the 6 Pro, I've not had any issues

I believe it's something specific to Google's OS.

eiginn, 2024-08-02:
I'm also experiencing this on a Pixel 8 Pro, fully updated. Happy to pull any logs or test builds if that's useful.

garthvh, 2024-08-03:
Android bellow 12 needs precise location permissions for Bluetooth.

andrekir, 2024-09-01:
there was an issue in 2.4.0 receiving the pairing response from the Android system. this was fixed in recent releases.

closed 2024-09-01 by andrekir