meshtastic/Meshtastic-Android #1173: [Bug]: Fixed Position Overwritten if "Share Phone Location With Mesh" Is Checked.

rjDipcord, 2024-07-31:
This issue seems to have been brought up before, and the previous complaints were closed without a real resolve.

This is not a duplicate of #1029. And I am not asking for the solution requested in #861, although that could be a option for this as I mention below.

The issue at hand is that when connecting to a node which has **Use Fixed Position** toggled _ON_ and with the app's **Share Phone Location With Mesh** toggled _ON_, the fixed position is overwritten by the phone's GPS coordinates.

On the surface, this sounds like preferred behavior. I can assure you _it is not._

Reasons this is an issue: 
1. Users who manage multiple nodes may not want the location of their node to be changed.
2. Users who connect to their nodes over WiFi might not be standing right next to their node, which makes the node's position inaccurate in relation to the android device.
3. Changing a node's location remotely is not possible due to 'spoofing' concerns. So it shouldn't be possible to change a 'fixed' node's location unless the user explicitly requests it.

"_But @rjdipcord, if someone doesn't want their phone's position shared with the mesh, they can just turn it off when connecting._"

Wrong. The **Share Phone Location With Mesh** toggle is _only_ available if you are already connected to a node. So, in a disconnected state, the user has _absolutely no way_ to disable location sharing unless they revoke location permissions from the app through the Android App Permissions system. Which is a problem that #861 was suggesting a solution for. If the toggle was adjustable even when disconnected from a node, then the above issue would be less of a problem.

With that said, it looks like we have a few potential solutions already:
1.  Make the **Provide phone location to mesh** toggle adjustable even when _disconnected_.
2. Make the **Use fixed position** flag gospel, and ignore provided phone location if this is toggled ON.
3. Make the **Provide phone location to mesh** toggle persistent only to the node it's switched on for. Not global to every node in the app.

This is a problem with release 2.4.0 alpha, and I have witnessed it as far back as 2.3.3. Device firmware also seems to have no affect on this problem. Currently on 2.4.0.46d7b82 and it behaves as it did in 2.3.8.

garthvh, 2024-08-01:
On both iOS and android this is an app setting, it is not stored on the device.

On recent firmware versions when fixed position is set it stays set at the same location until unset, once set to false a new position can be set.

ianmcorvidae, 2024-08-01:
I believe that Android is still sending phone position packets to the device, and in some cases those packets appear to be getting broadcast to the mesh. I'm not sure if anyone has dug into why that might be happening, but probably shouldn't be happening, in any case. The actual stored position that's sent out by the node automatically doesn't get overwritten, but in practice what that looks like is just that it "jumps back" to the stored fixed position when share phone position is turned off or the client is disconnected (plus however much time is needed until the node broadcasts again).

I think there's two things here:
a.) somehow allow "share phone location with mesh" to be toggled when disconnected (as mentioned it's only available when connected to a node -- I've gotten in the habit of connecting to something that doesn't send position publicly to check the setting is as I want it first, regardless of what node I actually want to use)
b.) chase down what's causing phone position packets to still make it to the mesh when fixed position is on

rjDipcord, 2024-08-01:
Thanks for the comment, @ianmcorvidae.
>The actual stored position that's sent out by the node automatically doesn't get overwritten, but in practice what that looks like is just that it "jumps back" to the stored fixed position when share phone position is turned off or the client is disconnected (plus however much time is needed until the node broadcasts again).0

It would seem you're correct. I just checked my node that exhibited the problem before, and it did in fact jump back to the stored position. I don't believe this is the behavior I witnessed in older versions.

garthvh, 2024-08-01:
iOS does the same thing and I am not seeing phone packets being broadcast with firmware that supports the updated fixed position, if it is that needs to be opened in the firmware repo.

closed 2024-08-01 by garthvh