meshtastic/Meshtastic-Android #1181: [Bug] Push notifications persist after messages are read

jeremiah-k, 2024-08-05:
After opening the app via a push notification to read a message, and message/channel is marked as read, the notification still needs to be manually swiped away.

madeofstown, 2024-08-05:
Been this way for as long as I can remember, would love to see this fixed.

jeremiah-k, 2024-08-05:
Yep. I believe that it's been like this since notifications were added. I've always noticed that it behaved differently than other messaging apps, but I finally remembered to check to see if there was an issue on it. 😄

Relatively minor bug, but if it were fixed I think it would make the user experience a little bit smoother.

andrekir, 2024-08-06:
I can't reproduce this issue. what Android version/device models are you running?

jamesarich, 2024-08-06:
> I can't reproduce this issue. what Android version/device models are you running?

Same - and we are supplying the `setAutoCancel` flag to the notification. https://github.com/meshtastic/Meshtastic-Android/blob/6d01fb4bc41b072903c4a2470bd78362b9474f3a/app/src/main/java/com/geeksville/mesh/service/MeshServiceNotifications.kt#L166

jeremiah-k, 2024-08-06:
> I can't reproduce this issue. what Android version/device models are you running?

I run  GrapheneOS (Android 14) w/ Sandboxed Play Services on a Pixel 6 Pro as my daily driver so that's where I notice it the most, but I thought this was more of a universal behavior so I haven't been paying close attention to how it performed on different phones.

I have a number of phones with different versions of stock AOSP and a few with custom roms. I should have some time to update everything to the latest release tonight or tomorrow.  I'll do some more testing at that time.

madeofstown, 2024-08-06:
> I run GrapheneOS (Android 14) w/ Sandboxed Play Services on a Pixel 6 Pro as my daily driver so that's where I notice it the most, but I thought this was more of a universal behavior so I haven't [...]

I too am running graphene OS on a pixel 7a. Also tested on an older Samsung tablet with Android 13.

madeofstown, 2024-08-08:
Just tested this again and it does in fact clear the message. But it doesn't take me to the channel where the message was received. Instead it takes me to the node list.

jeremiah-k, 2024-08-09:
> Just tested this again and it does in fact clear the message. 

Same. I can't get it to happen on any of my phones now either.  Every time I would be out range testing this summer I'd run into it and never remember to check for or file an issue.

> But it doesn't take me to the channel where the message was received. Instead it takes me to the node list.

Yes, that was part of what felt clumsy about it too. The notifications would stick and it would open to the channels list. It would be good to have it take you to the specific channel or DM, but maybe that should be a separate issue? Or this one could be renamed. Whatever's best.

As far as I'm concerned, this one can be closed. I  can't recreate the original issue at the moment (thankfully😌).

closed 2024-08-09 by garthvh