meshtastic/Meshtastic-Android #1188: Add Last Heard to Map

fti7, 2024-08-08:
Add "Last Heard" to Map View.

Either additionally to the already shown "Last Position" or with some toggling Options

jamesarich, 2024-08-09:
It's already pretty crowded in the  InfoWindow we're jamming multiple strings into the 3 available views: `title`, `snippet\description`, and `subDescription`

https://github.com/meshtastic/Meshtastic-Android/blob/6d01fb4bc41b072903c4a2470bd78362b9474f3a/app/src/main/java/com/geeksville/mesh/ui/map/MapFragment.kt#L240-L246

https://github.com/osmdroid/osmdroid/blob/osmdroid-parent-6.1.14/osmdroid-android/src/main/java/org/osmdroid/views/overlay/OverlayWithIW.java

Anything more in the current views would probably not fit, but there are opportunities to provide a new InfoWindow view.

b8b8, 2024-08-17:
how about colour code the info window background. Green for zero hops, lighter green for 1 hop, really light green for 2 hops. orange for 3 hops etc.

andrekir, 2024-08-18:
adding last heard along with position timestamps can be confusing. for now the map simply shows nodes and position data, and you can longpress the node marker to jump to it and view more details.