meshtastic/Meshtastic-Android #1193: Switching apps causes radio config screens to zero-out in-progress changes

scotty6435, 2024-08-12:
Whilst setting up my first node today I had an annoying problem. As I was setting it up with my phone and my phone had a picture of my SSID password, as I switched between apps to read the values out, the app would delete it.

I can understand why the app would play safe and pull down a fresh copy of the config when it comes back into the foreground but it would be a lot better if:

 * The UI tracked when a change was made anyway (colour and icon/text status)
 * On refresh anything that had been previously recorded as 'modified' was retained.

prokrypt, 2024-08-13:
What phone do you have and what are its specs? I suspect your phone is running out of memory (or doing some Weird Things) and unloading the app when you switch to another app.
I have no such issue with my phone, unless I load the camera which then kills everything in the background 🫠

scotty6435, 2024-08-13:
It's a pixel 7a so not a low spec phone. I was also live switching between two apps, there would be no time for the OS to suspend meshtastic

b8b8, 2024-08-18:
Interesting. I just tried this on my Pixel 7a and was able to switch between pictures the password box, and any other app.

scotty6435, 2024-08-18:
I'm not getting it either. I can only think that either it only manifests on a brand new device, via Bluetooth or my phone was somehow under such ram load that it was pushing the app into the background immediately

commit 160f0201 (2024-08-18): fix(config): use `rememberSaveable` to retain state across configuration changes

closes #1193
closed 2024-08-18 by andrekir