meshtastic/Meshtastic-Android #1196: [Feature request]  Show the hop count on the node pins on the map (like this https://pole1.co.uk/node-plot/ )

Paul-Pressure, 2024-08-16:
[Feature request]  
Show the  HOP COUNT  on each node pin on the map 
like this  https://pole1.co.uk/node-plot/

Paul-Pressure, 2024-08-16:
Just to add, I feel having this information will help as all get a better understanding of the real world routing for the good of the mesh :)

b8b8, 2024-08-17:
how about colour code the info window background. Green for zero hops, lighter green for 1 hop, really light green for 2 hops. orange for 3 hops etc.

andrekir, 2024-08-18:
the map is simple and for now not really intended for showing complex data or as a network topology tool. also keep in mind hops aren't static, especially in larger meshes, they can change a lot. for a more in-depth network visualization tool have a look at https://github.com/meshtastic/network-management-client

closed 2024-08-18 by andrekir

Paul-Pressure, 2024-08-18:
Hi @andrekir  no problem understand