meshtastic/Meshtastic-Android #1207: [Feature Request]: Rearrange traceroute result and add some differentiating colour

b8b8, 2024-08-24:
### Tell us your idea.

My suggestion is to have the traceroutes present vertically and with different colours for hops, transmit, and receive. Also including the hopping SNR would be beneficial.
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

GUVWAF, 2024-08-25:
Hey @b8b8, maybe it would be good to consider here how traceroute in 2.5 is going to look like in general. Apart from the SNR per hop, we can also get a two-way traceroute result. There is some validation needed, because it depends on the firmware of the nodes in the route whether the info is accurate or not. I opened a PR for the Python CLI (https://github.com/meshtastic/python/pull/654) to accommodate for this; basically the length of the list of SNRs should be 1 longer than the route list and nodes and SNR can be unknown (represented by `0xFFFFFFFF` and -128 respectively).

Maybe it can look something like this then (excuse me for my Paint skills):
[image]

Not sure how feasible all of this is, but it would be cool to show it like this :)

b8b8, 2024-08-25:
Dude this looks even better.

Nestpebble, 2024-08-25:
I've already mentioned it in the discord, but apparently not asked for it as a feature request. Putting the results in the chat history for a node would solve some of the things people have asked for above

b8b8, 2024-08-25:
> I've already mentioned it in the discord, but apparently not asked for it as a feature request. Putting the results in the chat history for a node would solve some of the things people have asked for above

yea, how about the pop-up, but then also the pop-up results end up saved in the appropriate DM with text of all the info people are asking for. Pop just shows info GUVWAF posted. DM has all this plus any other info like timing as text.

Nestpebble, 2024-08-25:
> yea, how about the pop-up, but then also the pop-up results end up saved in the appropriate DM with text of all the info people are asking for. Pop just shows info GUVWAF posted. DM has all this [...]

Yep. They're a bit ephemeral at the minute, even if they succeed after you switch tabs they're lost.

GUVWAF, 2024-08-25:
@cbvicious 

> Is there any possibility to add timestamps to each hop? 

No, those are not sent over LoRa. It would also be hard to get valuable info from them, because each node uses random delays before transmitting and they might have other packets in their queue. Also nodes are not time-synchronized, at least not with second-level accuracy. 

> And/or how many attempts it took to make each hop? 

Intermediate hops only ever try once. If the signal quality is strong and the traceroute fails, it's likely a packet collision.

andrekir, 2024-09-01:
the traceroute dialog message takes a `String` type, so more complex layouts are not possible. someone could add colors using `SpannableString` but that's a lot of work. priority for now is to add the new 2.5 round-trip/SNR information in.

closed 2025-05-24 by github-actions[bot]