meshtastic/Meshtastic-Android #1214: Position precision slider in metric

valzzu, 2024-08-28:
### Tell us your idea.

Would be nice if the position precision slider was in metric for us what use it :)

In mqtt and channel settings 

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

b8b8, 2024-08-28:
my suggestion that i posted on discord at the time was to just display both units. this is only shown when actually making the change and there seems to be sufficient room.

andrekir, 2024-08-29:
go to **Android Settings > Language** and change your Country to something other than ... (United States) or ... (United Kingdom)

closed 2024-08-29 by andrekir

mkgin, 2024-08-30:
> my suggestion that i posted on discord at the time was to just display both units. this is only shown when actually making the change and there seems to be sufficient room.

Both units would just be confusing... likely undesired by most... off to the shop to by some two by fours... which will surely be less than 5cm x 10cm.

b8b8, 2024-08-31:
Andre is right. I had the same issue. For some reason EVERYTHING on my phone was set to metric/Canadian except somehow language defaulted to USA. It works now perfectly. Idk if it was random android update.

valzzu, 2024-09-01:
ye now i got in metric :)