meshtastic/Meshtastic-Android #1217: [Feature Request]: Direct Message Notifications

BobsBlueNorth, 2024-08-29:
### Tell us your idea.

I am running the Meshtastic Android App V 2.4.4. with devices running firmware 2.4.2

I noted that Direct Messages no longer show up in the "Messages" screen with the other channels. It appears the user can only get to them by tapping the node in the Contact/Node List and tapping "Direct Message".

If this is an intentional feature besides an Android notification there is no way to tell that a user has an unread DM.  Any thoughts about putting an icon next to the node in the "Contact/Node List" to show an unread DM?

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Talie5in, 2024-08-30:
Just tested 2.4.4 App against 2.5.0TP2 firmware and I am not seeing issues described.

Go to messages tab, previous DMs there.
Chose new new to.message from node list to message, send message, leave message window, go to node list, map view, back to messages, message to new node shows in list.

BobsBlueNorth, 2024-08-30:
Yup. I went back and played with it some more. 

It appeared that when I switched between devices without closing the Android app it caused this 'bug'.  As soon as I closed and reopened the app all was good to go.

andrekir, 2024-09-11:
fixed in https://github.com/meshtastic/Meshtastic-Android/commit/41fc43b2150f4857c1d08511a5378b80752e4bd0

closed 2024-09-11 by andrekir