meshtastic/Meshtastic-Android #1218: Wrong role displayed on device list (client instead of Router)

Mertszr02, 2024-08-30:
### What happened?

Device list shows shows all routers as clients, 
I have 2 routers and 1 old router_client.
One of the routers (R2) is near me so its more up to date v2.4.1 
I don't remember the versions of the other two.
List shows my own client_mute device correctly.

[image]
[image]
[image]

### App Version

2.4.4

### Phone

Xiaomi 13T

### Device

Rak4631

### Firmware

2.4.1

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Mertszr02, 2024-08-30:
Ok this is interesting. I have changed nothing yet the nodes list now shows the correct roles.
My node was in my room disconnected from the phone, I came back and checked the list.

Could it be:
Role of the remote nodes were unknown at first and the default value is client so it shows client?

[image]

jamesarich, 2024-08-30:
> Role of the remote nodes were unknown at first and the default value is client so it shows client?

Yep, this is indeed the case - the default (unset) value of 0 equates to `CLIENT` role.

https://github.com/meshtastic/Meshtastic-Android/blob/cb2ea2804a19caec29f0b8322cf26872e21904a5/app/src/main/java/com/geeksville/mesh/NodeInfo.kt#L32-L33

Mertszr02, 2024-08-30:
Thanks, this feels kinda misleading. Not a big deal but I think it can have a default (eg: unknown) value

jamesarich, 2024-08-30:
#1219 adds defaults for the device values in that row when unknown

PR #1219 (2024-09-16): feat: adds UI for unknown node details
This commit provides default device details shown in the NodeInfo cards, providing visual clarity when the information is unknown.
[image]

resolves #1218
closed 2024-09-16 by andrekir