meshtastic/Meshtastic-Android #1220: Direct message name mixup

Mertszr02, 2024-09-01:
### What happened?

Direct messages are mixed up, when I send a message to a nearby node the name shows up wrong(shows another node[9km and 3 hops away)
This behavior is not new, I have seen it on older firmware and app versions. But the problem always seem to happen on wisblock devices, seems to me like the app somehow mixes them (as if they have the same identifier)

Has anyone seen this behavior?

https://github.com/user-attachments/assets/af05d09d-260a-4999-9148-567999b73eb9

### App Version

2.4.4

### Phone

Xiaomi 13T

### Device

Rak4631

### Firmware

2.4.2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Mertszr02, 2024-09-01:
Don't know if this is relevant but I realized the way I worded it sounds like a node must be close by. This happenes randomly whether or not other nodes are close.

GUVWAF, 2024-09-01:
Yes, probably caused by nodes having the same NodeNum. RAK shipped a batch of devices with an exact copy of the firmware including config, causing them to have the same NodeNum. If you do a factory reset (check "Full Erase and Install" in the Web Flasher), it should be resolved.

closed 2024-09-08 by andrekir