meshtastic/Meshtastic-Android #1237: [Feature Request]: Default Sharing Channels - Exclude admin channel

ADADarkpool, 2024-09-10:
### Tell us your idea.

I'd like the app to...

Default to exclude a matching "admin" channel when creating QR codes / sharing channel information.

This would be a QOL feature.  I only need to share the admin between my own personal devices.  Because its a potentially a security issue, the friction of an extra click / selection should be on the enablement side and not on the default setting.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Kealper, 2024-09-10:
Adding to this: The `gpio` special channel (used by the RemoteHardware module for remote GPIO control on a node) could use the same exclusion by default. I think those two are the only special channel names in-use at the moment.

andrekir, 2024-09-11:
2.5 will use asymmetric encryption for remote admin (same as DMs) so admin channels will be gone soon

closed 2024-09-11 by andrekir