meshtastic/Meshtastic-Android #1239: [Feature Request]: Add DOP Circle to map for connected node

b8b8, 2024-09-10:
### Tell us your idea.

I think it would be a good idea to add the Horizontal DOP to the android app map as just a simple thin black circle. This is already something that can be included in the position packets, so it makes sense to show it. It can also be shown for all other nodes though I think in that case it will make the map too hectic since other nodes already have their precision circles shown. We are getting some nice and robust GPS code recently along with many hardware GPS now supporting multiple constellations and multiple frequencies so it would be beneficial to show it.

https://en.wikipedia.org/wiki/Dilution_of_precision_(navigation)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jamesarich, 2025-03-12:
@b8b8 has this been resolved by #1177 ?

jamesarich, 2025-03-12:
oh - rereading this is in addition to precision circles.

b8b8, 2025-03-12:
> oh - rereading this is in addition to precision circles.

Correct, it's for a single thin lined GPS DOP circle (that can be toggled on and off)

garthvh, 2025-03-13:
This data is pretty hard to interpret, it is not an accuracy in meters like the os positions provide. I have never been able to use it for anything.

b8b8, 2025-03-13:
> This data is pretty hard to interpret, it is not an accuracy in meters like the os positions provide. I have never been able to use it for anything.
> 
> The computation section on Wikipedia demonstrates all the necessary computations to make it useful.

Can i ask then why both DOP and HVDOP are selectable to send with a position packet?

garthvh, 2025-03-13:
It was added while the project did not really have a leadof by a gps interested contributor.

garthvh, 2025-03-13:
https://github.com/meshtastic/firmware/issues/3984

jamesarich, 2025-03-14:
closing this, doesn't sound like we can use this as originally proposed.

closed 2025-03-14 by jamesarich