meshtastic/Meshtastic-Android #1249: [Feature Request]: Change timezone setting from a textbox into a dropdown list

battlehax, 2024-09-16:
### Tell us your idea.

I believe a useful and hopefully simple enhancement would be to change the tzdef setting in **Radio Config>Device>POSIX Timezone** into a list containing the zones listed in the link from the docs, https://github.com/nayarsystems/posix_tz_db/blob/master/zones.csv
The list could also possibly display the human readable timezone from the first column of the csv, and apply the proper posix timezone from the second column.
Not a huge deal, but it would be nice, and save some setup steps.
Would this be possible/beneficial?

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

madeofstown, 2024-10-04:
I think that's way too many options for a drop down but I don't know of a better way to do it. As it is now, it's annoying to have to lookup and type in the proper code but I think it would also be annoying to scroll through all the different timzone options.

garthvh, 2024-10-04:
You can detect locale and sett the right value, drop down would be awful and most users would not know what to pick

Firecul, 2025-01-02:
What about just a standard UTC(+/-) box, that might be simpler and more obvious as to how to enter the correct format?

todd2982, 2025-03-27:
> What about just a standard UTC(+/-) box, that might be simpler and more obvious as to how to enter the correct format?

It's a great idea until DST comes back around.

closed 2025-05-24 by github-actions[bot]

brebtatv, 2025-06-24:
The automatic detection wouldn't be bad. I like that.