meshtastic/Meshtastic-Android #1265: [Bug]: Fixed Location,  I could not get this to set via the app after update to 2.4.12

Paul-Pressure, 2024-09-21:
### Contact Details

paul.zzard@live.co.uk

### What happened?

Fixed Location,  I could not get this to set via the app I had to resort to CLI  
This may well be relates to Firmware issue as a lot of updates.  
Most of us have multiple nodes, a fixed location homebase, and a couple that need to get location from the smart phone whilst mobile but fixed at home.   I noticed when I go into position setting it overwrites  my set fixed location.   These need to be locked in stone if possible.   Looking at my node list after a recent reset out of 22 nodes  7  have lost their location settings  that usually have them.  Thanks for all your hard work. 

### App Version

2.4.12

### Phone

galaxy s23 Ultra

### Device

Heltec V3

### Firmware

2.5

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

closed 2024-09-28 by andrekir

andrekir, 2024-09-28:
this should be fixed in release [2.4.15](https://github.com/meshtastic/Meshtastic-Android/releases/tag/2.4.15) if you find any issues please comment and we'll reopen.

glitchphan, 2024-12-02:
Hello, I am not sure if this is related.. I'm on app 2.5.7 with a Heltec V3 on 2.5.11. I tried turning on Fixed Position, entered coordinates, and for a very short while the app showed me at the provided ones. Then suddenly it changed to a neighbor node's coordinates instead. Tried this multiple times with same results.

glitchphan, 2024-12-03:
Just tested tried and got the same behavior on a rak4631 node that has a GPS module but is set to disabled. Now both nodes are on firmware 2.5.15.

commit 96a69b73 (2024-09-28): fix: position handling when `fixed_position` is enabled

closes #1265