meshtastic/Meshtastic-Android #1272: [Feature Request]: Cached settings for admin commands

jeremiah-k, 2024-09-27:
### Contact Details

Jack! on Discord

### Tell us your idea.

It would be really helpful if the app could cache a remote node’s admin settings so we don’t have to request them each time we want to make a change.

A lot of times it’s unnecessary to fetch the config from a remote node each time when we're the only one controlling it. If we’ve already accessed their settings, caching them would make things faster and would help reduce strain on the mesh.

For example, when I’m adjusting LongFast's location precision for a tree node, I have to re-request the channel settings before making each adjustment. Being able to use a cached versions of the settings would make this type of process a lot smoother.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

andrekir, 2024-09-27:
> For example, when I’m adjusting LongFast's location precision for a tree node, I have to re-request the channel settings before making each adjustment. 

you don't have to re-request the channels after each adjustment. after fetching the channels you can make multiple changes and send them one-by-one or all at once.

jeremiah-k, 2024-09-27:
> you don't have to re-request the channels after each adjustment. after fetching the channels you can make multiple changes and send them one-by-one or all at once.

That is true. In that instance, if using a second phone & device you can use that to check the changes on the map before backing out of the menu completely on the admin device.

If this isn't a direction that you're interested in going, that's okay.  It was just something I ran into recently and thought it might be be worth considering anyway.

andrekir, 2024-09-27:
I can totally relate! skipping the config fetch would be convenient but adds the risk of getting data out of sync. also 2.5 introduced a 5 minute window for changes after we fecth any config to prevent replay attacks, so there really isn't room for optimization.

closed 2024-09-27 by andrekir