meshtastic/Meshtastic-Android #1273: [Bug]: Number of satellites does not always populate for connected node

b8b8, 2024-09-28:
### What happened?

The icon under the voltage/battery percentage does not always populate with the number of satellites the node is connected to. Viewing the same node from another device correctly shows the GPS is connected to said satellites. 

Maybe an update as well that shows a flashing icon showing the GPS hardware is detected (but not signal locked) and signal strength like the device LCD screen has next to the number of connected satellites would be helpful for (in this case) screenless devices.

### App Version

2.3.14 and 2.4.15

### Phone

Pixel 7a

### Device

T1000-E

### Firmware

2.5.2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

andrekir, 2024-09-28:
> The icon under the voltage/battery percentage does not always populate with the number of satellites the node is connected to. Viewing the same node from another device correctly shows the GPS is connected to said satellites.

if the data is from the radio, it will show the number of satellites. however if Android location data is used, it doesn't have any satellite info. 

> Maybe an update as well that shows a flashing icon showing the GPS hardware is detected (but not signal locked) and signal strength like the device LCD screen has next to the number of connected satellites would be helpful for (in this case) screenless devices.

clients do not have access to that information

b8b8, 2024-09-28:
> if the data is from the radio, it will show the number of satellites. however if Android location data is used, it doesn't have any satellite info.<

Correct, the location sharing from android is not enabled, but still not satellite info is shown.

andrekir, 2024-09-29:
thanks for the detailed information. does this happen right after connecting to the radio?

nodes store in the nodeDB a [trimmed-down version](https://github.com/meshtastic/protobufs/blob/83c78e26e39031ae1c17ba5e50d0898842719c7f/meshtastic/deviceonly.proto#L23) of the Position protobuf to save space, which doesn't include satellites in view. so clients will only receive sats in view from periodic broadcasts.

thats why right after connecting, satellite info for every node will be missing (unless there are Position packets in the toPhone queue).

b8b8, 2024-09-29:
Yes there are no satellites seen right after connecting (which is understandable), but also the same lack of satellites hours later. 
In the screen shot below I used my connected to Seeed nodes to request the position of my WX node. The number of satellites instantly updated to 7 however as you can see no satellites are shown for the connected Seeed node even though it should be polling the GPS for a new position every 3min.

[image]

[image]

andrekir, 2024-09-29:
> In the screen shot below I used my connected to Seeed nodes to request the position of my WX node. The number of satellites instantly updated to 7 however as you can see no satellites are shown for the connected Seeed node even though it should be polling the GPS for a new position every 3min end when requesting the WX position it should be sending its own position.

position packets (including both requests for position and periodic broadcasts) are all handled at the same place in the code. missing satellite info is likely due to the packet not including satellite data (satsInView = 0).

b8b8, 2024-09-29:
It is included. And the number of sats seen by the seeed device is shown on remote devices, just not the connected device. 

[image]

andrekir, 2024-09-29:
not included in the position packets received from your node. it’s possible that your node is sending satsInView = 0.

b8b8, 2024-09-29:
then how is my other node receiving the number of satellites correctly from this same node?

b8b8, 2024-10-04:
i've done some more testing and the satellites seem to show more often when driving around, even with the GPS refresh rate set to 180 seconds. My whole drive home today i could see the satellites of the connected device, however at home and stationary i could not see the satellites of the connected device.

andrekir, 2024-11-14:
@b8b8 Check your position log: do you have any positions showing 0 Sats? That's the raw position packet coming from the radio.

b8b8, 2024-11-15:
I do have the odd zero satellites which could explain it then. I know now for whatever reason this is working FAR MORE reliably now. I think a bunch of updates to the GPS code and getting the Seeeed tracker GPS working correctly has essentially solved this issue. You can close this if you want now. thanks for the effort trying to track this down.

closed 2024-11-15 by b8b8