meshtastic/Meshtastic-Android #1288: [Bug]: Device Telemetry - Power metrics: current values

joneyes, 2024-10-04:
### Contact Details

jon.eyes@duncreggan.net

### What happened?

Mutliple current values displayed - only started on this version.
node has a single channel INA 219 current sensor used to monitor battery charge/discharge current.
The same symptom occurs when data is received via the mesh, or directly connected to the RAK via BLE
[image]

### App Version

2.4.17

### Phone

Samsung s22 Ultra, Android 15, One UI 6.1

### Device

RAK4631, with INA219 current sensor board connected on I2C

### Firmware

2.5.3.a70d5ee

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

andrekir, 2024-10-04:
all data comes from the radio. power metrics channel 2 & 3 were added in .17 and the app displays that data as received.

joneyes, 2024-10-05:
Hmm, OK.
I have the following observations then.

There is a single ina219 sensor connected, with a single channel.
It's connected low side.
Prior to this change, the measured idle current was about 11ma, which from memory, reflected what the board was drawing when measured with a multimeter.
I now see two current values, neither of which reflect the load current, either individually or as a sum?

b8b8, 2024-10-06:
Try bridging A0 to hard set the i2c address. I had to recently do this to an INA3221 to get it to read correctly. Sometimes the default address isnt picked up correctly.
[image]

joneyes, 2024-10-10:
I looked at the debug.
The INA219 is correctly identified in the intialisation:

_INFO | ??:??:?? 4 INA219 sensor found at address 0x40
INFO | ??:??:?? 4 1 I2C devices found
DEBUG | ??:??:?? 4 acc_info = 0
DEBUG | ??:??:?? 4 found i2c sensor meshtastic_TelemetrySensorType_INA219_

I can also see the data being logged as sent:
_INFO | ??:??:?? 18 [EnvironmentTelemetryModule] Environment Telemetry: Initializing
INFO | ??:??:?? 18 [EnvironmentTelemetryModule] Init sensor: INA219
INFO | ??:??:?? 18 [EnvironmentTelemetryModule] Opened INA219 sensor on i2c bus
INFO | ??:??:?? 18 [PowerTelemetryModule] Power Telemetry: Initializing
INFO | ??:??:?? 19 [EnvironmentTelemetryModule] (Sending): barometric_pressure=0.000000, current=169.699997, gas_resistance=0.000000, relative_humidity=0.000000, temperature=0.000000
INFO | ??:??:?? 19 [EnvironmentTelemetryModule] (Sending): voltage=0.000000, IAQ=0, distance=0.000000, lux=0.000000
INFO | ??:??:?? 19 [EnvironmentTelemetryModule] (Sending): wind speed=0.000000m/s, direction=0 degrees, weight=0.000000kg
DEBUG | ??:??:?? 19 [EnvironmentTelemetryModule] Initial packet id 87601972
DEBUG | ??:??:?? 19 [EnvironmentTelemetryModule] Partially randomized packet id 2250073909
INFO | ??:??:?? 19 [EnvironmentTelemetryModule] Sending packet to mesh_

I only see one current value in the telemetry?
This resulted in the following being received in the app (now version 2.5.0)
[image]

brebtatv, 2024-10-16:
I have a similar problem. In addition, my INA219 has the option of addresses just like a colleague. This makes it impossible to use three INA219s. According to the documentation, other addresses are different.

b8b8, 2024-10-17:
> I have a similar problem. In addition, my INA219 has the option of addresses just like a colleague. This makes it impossible to use three INA219s. According to the documentation, other addresses are different.

Use an INA3221 if you need 3 power readings. I have one monitoring load, solar in, and battery. It works great.

brebtatv, 2024-10-23:
[image]
There are more of these problems. Current and voltage in the environment? At the same time in Power.

joneyes, 2024-10-23:
Also, current reading is displayed in the incorrect units. These values are actually mA.

[image]

b8b8, 2024-10-23:
Yup, this stuff was mention already in discord. Remember this is alpha release. Amps to mA should be fixed next release.

andrekir, 2024-10-27:
all power metrics Telemetry data now moved to the node details screen

closed 2024-10-27 by andrekir