meshtastic/Meshtastic-Android #1290: [Feature Request]: Display LocalStats

GUVWAF, 2024-10-05:
### Tell us your idea.

The device now sends its LocalStats over the PhoneAPI every 15 min. It includes statistics like the amount of good and bad packets received, amount of transmitted packets and, more recently (https://github.com/meshtastic/protobufs/pull/591), the amount of duplicate received messages, amount of transmitted relayed packets and canceled relayed packets. As mentioned in #1200, it thus gives some more raw data from the device that's otherwise not available.
It would be nice to show this somewhere in the Android app, maybe on the device connection page (like iOS), but I'm unsure what the best place would be.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Kealper, 2024-10-10:
I feel like the empty spot below the option to provide phone location to the current device would be a great place for this extra info. It's an empty area that's already available and otherwise unused, so it makes sense to put that sort of local device specific data in there.

b8b8, 2024-10-12:
That area is not actually existent if you have many mesh devices on your list since the scrolling makes nodes disappear at the top.
[image]

garthvh, 2024-10-12:
The data is designed for widgets, long duration for things like always on screens.

Kealper, 2024-10-12:
> That area is not actually existent if you have many mesh devices on your list since the scrolling makes nodes disappear at the top. But rejigging the list to force a separate area down there could work. ![image](https://private-user-images.githubusercontent.com/156552149/375921008-763d7e4c-8650-426e-93bc-4b864f47c2f4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg3MDcyNDIsIm5iZiI6MTcyODcwNjk0MiwicGF0aCI6Ii8xNTY1NTIxNDkvMzc1OTIxMDA4LTc2M2Q3ZTRjLTg2NTAtNDI2ZS05M2JjLTRiODY0ZjQ3YzJmNC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAxMlQwNDIyMjJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jM2Y2MDI1NDU1MTljYTg4MDdkOTRkODI0MzBiMDM5ZGEyYmE0Y2Q4YTk1MmQyY2YzMzY5ZjA1MGQ4NmJmZjgyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.j_7UPF-LwCESfvFesi4TCewbuqH5KitwYwHikbHNxHs)

Clearly, I need to up my game! 🤣

PR #1307 (2024-10-13): feat: Add local device stats to the service notification
This adds the `Telemetry.LocalStats` to the service notification. This information includes the number of online and total nodes, as well as other local stats. It also updates the notification summary and adds local stats telemetry handling.

This implementation should also automatically display new fields in the proto as they are added.

resolves #1290 
[image]
[image]
closed 2024-10-13 by andrekir