meshtastic/Meshtastic-Android #1310: [Bug]: Error: Unknown Public Key

joshbowyer, 2024-10-12:
### What happened?

Trying to direct message between two devices, fresh factory reset and nodes deleted from app on both sides. Node A sees Node B in node list, Node B never sees Node A. Trying to direct message from Node A to Node B fails quickly and tapping on the cloud/slash icon says Error: Unknown Public Key. Both nodes are on 2.5.5 and the app is 2.5.0 (Android)

### App Version

2.5.0

### Phone

Pixel 4, Samsung S22

### Device

RAK4631, Waveshare LoRa hat/RPi

### Firmware

2.5.5

### Relevant log output

```shell
On Node B, debug panel never shows any packet to or from Node A.

On Node A, after attempting to send a direct message to Node B, it shows the packet to Node B with the correct nodeid, and the payload (message). However, the next packet says from Node B to Node A (reply), and payload is "\030#" (not sure what that means).
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mtsdawson, 2024-10-13:
I'm also having this issue. Both nodes are on 2.5.6 and android app is 2.5.1.

The public key from Node A is entered correctly in Node B -> Security -> Admin Key -> 1/3. 

I've completed a NodeDB reset on both. Issue persists with the same error message.

andrekir, 2024-10-14:
> Node A sees Node B in node list, Node B never sees Node A. Trying to direct message from Node A to Node B fails quickly and tapping on the cloud/slash icon says Error: Unknown Public Key.

Node B needs Node A's public key for PKC to work. this is handled in the firmware, not in the app. make sure your channels are setup correctly, let them discover each other and it should work.

closed 2024-10-14 by andrekir

joshbowyer, 2024-10-14:
Why was this closed? My nodes have the same channels, shared and configured via QR code. Yet I get the error, however the Android app shows a green lock (basically says it is secured by PKI but then when actually trying to DM says unknown public key).

andrekir, 2024-10-14:
if node B is doesn't show node A in the node list it won't work. that is expected. and this is not an app issue.

d3vv3, 2024-10-15:
Say you want to manage Node B with Node A (A controls B):

In node B `Radio configuration > Security > Admin Key` you have to place the public key of the node A. The Web UI provides a bit more information on this:

[image]

kalestew, 2024-10-21:
I am also experiencing this problem with the PKI DMs. 

I do not think this issue should be closed. At the very least a method to trigger key exchange manually should be a priority for improvement. 

@andrekir is right that the expected functionality is the devices should update the keys automatically through the configured channel. This does not always happen. As @joshbowyer also observed, the issues with key mismatches are not always resolved with a nodedb reset and waiting a few hours with the nodes together. 

I will collect some logs and the keys from each node and report later when I have some time to test again.

andrewheadricke, 2024-10-28:
Also just got this problem. Please re-open the issue.

One node on 2.5.7 and another and 2.4.8, direct messages was working fine for several days. Now the 2.4.8 node cannot send messages anymore, gets 'unknown public key' error.

mtsdawson, 2024-10-28:
I've had this a few times between firmware revisions, likely do to a public key change. Fix is to delete the node or do a full NodeBD reset. Error goes away after a new handshake / updated key exchange.

brentgb, 2025-01-21:
I have this issue as well. Deleting the node, doing a full NodeDB rest, and flashing again does not fix it.

mkyung, 2025-06-22:
Had the same issue with 2.6.8. I have two GAT562. Both have the same channels by cloning via QR code. Fixed by resetting the Node DB.

The first channel is LongFast default public channel; Second channel is custom private. Never had an issue when chatting in the private channel, but device A cannot send to device B due to the `unknown public IP` even on the device A node list I was able to view the device B.

What I did was to reset the Node DB on both devices, then they rebooted themselves. 
Next restart the iOS app, try sending a message on the private channel. After few minutes the screen showing `@2/2` which means they both recognized each other on the node list. At this moment Im able to send DM messages.