meshtastic/Meshtastic-Android #1322: [Bug]: Remote admin: "Encrypted Send Failed" when "Licensed amateur radio" = True

nikvostro, 2024-10-16:
### Contact Details

@nikvostro

### What happened?

When trying to browse any section of remote node AND **Licensed amateur radio** is Enabled - error below appears.
If **Licensed amateur radio** is Disabled on admin node then access to remote admin parameters works fine (=> PKI keys are setup correctly) for any of my Meshtastic devices.
Troublshooting steps performed:
- Devices factory reset
- Flashed FW from the scratch (not update)
- Meshtastic android application data cleared
Error message appears immediately without any delays right after clicking on any section.
[image]

### App Version

2.5.1 (from Google Play)

### Phone

Samsung Galaxy S9, Android 10

### Device

3x LiliGO [T3-S3 SX1262](https://www.lilygo.cc/products/lora-v1-3) devices and one [T-Beam](https://www.lilygo.cc/products/t-beam-v1-1-esp32-lora-module) device

### Firmware

2.5.5 beta

### Relevant log output

```shell
Packet 10/16/24 22:55:55
from: SENDER (![0xSENDER]) (![0xSENDER])
to: SENDER
decoded {
  portnum: ROUTING_APP
  payload: "\030\""
  request_id: 1969081227
}
id: 3227049786
rx_time: 1729108554
hop_limit: 3
priority: ACK

Packet 10/16/24 22:55:54
to: RECIPIENT (!00000000)  <- ID is displayed broken, looks like: "xxxxxxx (!00000000)x" - where "(!00000000)" block separates last decimal digit of ID
decoded {
  portnum: ADMIN_APP
  payload: "\030\001"
  want_response: true
}
id: 1969081227
hop_limit: 3
want_ack: true
priority: RELIABLE
public_key: "[ENCODED_KEY_HERE]"
pki_encrypted: true
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

andrekir, 2024-10-16:
there is no remote admin for ham mode since there is no encryption.

closed 2024-10-16 by andrekir

nikvostro, 2024-10-16:
> there is no remote admin for ham mode since there is no encryption.

OK, now understand, thank you!

What about to replace error message text with something more detailed?
Like: "HAM mode doesn't allow encryption - remote admin is only possible when User:Licensed amateur radio is Disabled."

andrekir, 2024-10-17:
the error is generic and not only used when ham mode is the issue.

turnrye, 2024-11-05:
Hi @andrekir ! Another user surprised/confused by this. Would you be open to a PR that improved the user experience around this situation? I think adding some UX constraints, like disabling the related buttons and adding tips explaining the restriction, would be helpful.

The current state isn't intuitive or obvious because in many regulatory domains, ham radio users are legally allowed to use encryption. If anything I'd expect the system to be more permissive of me as a licensed operator rather than restrictive.

andrekir, 2024-11-05:
Users have all the tools to do whatever they want. It's very simple. If you want encryption, don't enable `is_licensed`.

UX can improve for sure, I started adding descriptions ("summary") to `DropDownPreference` in 2370e8b596c5f6f0c2d0edc80a57565eb73b190d, feel free to expand that to other Preferences or add some dialogs if you think it’d help make things clearer.