meshtastic/Meshtastic-Android #1325: [Feature Request]: Add pressure to Env Metrics and made selectable

GMart, 2024-10-17:
### Contact Details

Discord Droidbot2

### Tell us your idea.

I'd like the metrics screen to show the pressure recorded, if the node has a pressure reading on it. 
If the Env. Metrics screen gets too cluttered, than I also propose adding the ability to click the legend to toggle, so for example click IAQ in the legend and the IAQ graph would toggle between shown and hidden.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

b8b8, 2024-10-17:
I think ultimately the plan is to move VOC to its own "hazards" graph along with CO2 and later air-particulate. This would add room for pressure to be added to the temp and humidity graph.

b8b8, 2024-10-17:
https://github.com/meshtastic/Meshtastic-Android/discussions/1175#discussioncomment-10301430

closed 2025-05-24 by github-actions[bot]