meshtastic/Meshtastic-Android #1333: [Feature Request]: mark as read

cracky22, 2024-10-20:
### Contact Details

martinblieninger2208@gmail.com

### Tell us your idea.

Option that you can mark notifications as read by clicking in the notification center and thus the person opposite gets the read symbol

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

andrekir, 2024-10-20:
this is not possible without firmware support.

closed 2024-10-20 by andrekir

caveman99, 2024-10-20:
@andrekir can't you do something similar with tapbacks?

cracky22, 2024-10-20:
> this is not possible without firmware support.

Is it not possible to add this anyway or to discuss it with the firmware team?

andrekir, 2024-10-21:
@caveman99 a tapback sends a message reply. For a 'read receipt,' something different would need to be implemented.

caveman99, 2024-10-21:
Yes, I thought a special value for the tapback could be used. It requires a message back anyway, and these were already there.