meshtastic/Meshtastic-Android #1334: [Feature Request]: settings for notification

cracky22, 2024-10-20:
### Contact Details

martinblieninger2208@gmail.com

### Tell us your idea.

Setting option so that you can choose which pop-ups you get (just notifications or whether connected and how many users are online

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

andrekir, 2024-10-20:
You can already do this through Android’s **Settings > Apps > Meshtastic > Notifications**. There are currently three separate notification channels: service, message, and new node notifications. Choose which ones you want to see or customize how they appear.

Samsung hides that option in the Advanced settings, search for something like "Manage notification categories for each app".

closed 2024-10-20 by andrekir

hb9eue, 2024-10-27:
Thank you for telling how to disable, the 'new node' notification, which was driving me crazy!