meshtastic/Meshtastic-Android #1338: [Bug]: Android app not syncing time after latest update

thefoeyouknow, 2024-10-21:
### Contact Details

thefoeyouknow@gmail.com

### What happened?

Several days ago my android app updated. After that, when connecting to my t-watch s3, the clock now syncs only to 06:00. 
I've verified my posix tz is still correct, I've reflashed and reconfigured the node, but all I get is 06:00 at sync connection.

### App Version

2.5.3

### Phone

Pixel 7, android 14

### Device

Lilygo T-Watch S3

### Firmware

meshtastic v2.5.7.f77c87d

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

andrekir, 2024-10-26:
The way the app sends time on connect hasn't changed since 2.5.0 when it was first implemented. Can you grab some serial logs from the T-Watch?

thefoeyouknow, 2024-10-26:
I don't know how to do that, but I'll find some spare time to learn it today, and get logs for you.

thefoeyouknow, 2024-10-26:
[image]
Here's the serial console while the phone was connecting to the node.  Still trying to figure out how to export logs.

andrekir, 2024-10-26:
Thanks for the detailed information. Is the T-Watch connected to WiFi? If so, it automatically syncs time from the Internet using NTP and ignores any time sync from the app. This means the app has no effect on the watch's time at all.

thefoeyouknow, 2024-10-26:
Wifi puts the watch in a bootloop

andrekir, 2024-10-27:
Your watch thinks it has NTP time, which should only happen when connected to WiFi. The app is working as intended.

If you are not using WiFi and your watch thinks it has NTP time, you can open an issue in the firmware repo: https://github.com/meshtastic/firmware/issues

closed 2024-10-27 by andrekir

thefoeyouknow, 2024-10-27:
The watch has been moved around to multiple different firmware, multiple android devices have been tried, and the clock always sets to 0600 when the time zone is set, and 1200 when the config is clean.  
At this point, it has to be hardware somehow.