meshtastic/Meshtastic-Android #1347: Reopened: Bluetooth reconnect

DerWossi, 2024-10-25:
I could not re-open issue #913...

Same unsolved issue here.
I am using a LILYGO T-Echo with the latest firmware 2.4.2 connected to a Google Pixel 7 with Android 15.
The pairing is not a problem cause this is stable and i do not need to re-pair the device.
Just the app needs to re-connect as @Ztrby already wrote...

_Originally posted by @DerWossi in https://github.com/meshtastic/Meshtastic-Android/issues/913#issuecomment-2437046061_

Baadvo, 2024-10-25:
I have the same problem. 

Mobile: Android 12 Moto G50.
Mesh device:
- 1x T-Beam - FW2.5.5
- 2x WSL V3 - FW2.5.5

I'm pretty sure everything worked fine with the older FW. I'll test it when I get home from work.

DerWossi, 2024-10-25:
Update:
I just installed a T-Beam with tbeam 2.5.6.d55c08d -> same issue

caveman99, 2024-10-25:
are you sure you don't have 2 phones 'competing' for the device? Also try forgetting it from Android system dialogue. What's the debug log say about this?

DerWossi, 2024-10-25:
> are you sure you don't have 2 phones 'competing' for the device? Also try forgetting it from Android system dialogue. What's the debug log say about this?

Absolutly sure. The T-Beam by the way wasn't installed anywhere before. It came straight from my spare parts box.
I already removed and reestablishd pairing to the phone -> did not help.
Whitch log entries do you need?

Baadvo, 2024-10-25:
Upodate:

I tested it:
WSL V3 - **FW2.5.5** vs Google Play "Beta" version **2.5.3** - this combination does **not reconnect** after losing the connection.

WSL V3 - **FW2.4.5** vs github APK (https://github.com/meshtastic/Meshtastic-Android/releases/tag/**2.3.15**) - both combinations **work great**. The device instantly reconnects when I come closer.

@DerWossi please test it.

DerWossi, 2024-10-25:
I'll give it a try.
thx

Baadvo, 2024-10-25:
UPDATE:

I have discovered that FW4.5.6 sometimes works.

Log when it works:
```
... <- Some logging
[0m [32m INFO  [0m| 16:44:41 366 [32m BLE disconnect
[0m [32m INFO  [0m| 16:44:41 366 [32m PhoneAPI::close()
[0m [32m INFO  [0m| 16:44:48 373 [32m BLE authentication complete
[0m [32m INFO  [0m| 16:44:50 375 [32m To Radio onwrite
[0m [34m DEBUG [0m| 16:44:50 375 [34m New ToRadio packet
[0m [32m INFO  [0m| 16:44:50 375 [32m Client wants config, nonce=31
[0m [34m DEBUG [0m| 16:44:51 376 [34m Got 17 files in manifest
[0m [32m INFO  [0m| 16:44:51 376 [32m Starting API client config
... <- Sending configuration to the phone
[0m [32m INFO  [0m| 16:44:57 382 [32m To Radio onwrite
[0m [34m DEBUG [0m| 16:44:57 382 [34m New ToRadio packet
[0m [34m DEBUG [0m| 16:44:57 382 [34m PACKET FROM PHONE (id=0x61f5b086 fr=0x00 to=0x10, WantAck=1, HopLim=3 Ch=0x0 Portnum=6 priority=70)
[0m [34m DEBUG [0m| 16:44:57 382 [34m Enqueued local (id=0x61f5b086 fr=0x00 to=0x10, WantAck=1, HopLim=3 Ch=0x0 Portnum=6 rxtime=1729874697 priority=70)
[0m [32m INFO  [0m| 16:44:57 382 [32m Telling client we have new packets 16
[0m [32m INFO  [0m| 16:44:57 382 [32m BLE notify fromNum
... <- Normal communication
```

Log when it doesn't work:
```
... <- Some logging
[0m [32m INFO  [0m| 16:41:59 204 [32m BLE disconnect
[0m [32m INFO  [0m| 16:41:59 204 [32m PhoneAPI::close()
... <- Some logging
[0m [32m INFO  [0m| 16:42:21 226 [DeviceTelemetryModule] [32m Sending packet to phone 
... <- Some logging (The mobile never connects itself)
```
When I'm away long enough to do the "Send packet to phone", the phone no longer connects.

I'm gonna go try some other things.

Baadvo, 2024-10-25:
Even with FW2.4.2, the phone doesn't seem to connect when you are away for a long time.

I'm at the beginning of my journey. :-(

DerWossi, 2024-10-28:
I did a long time test this weekend. All the tested versions did only reconnect sometimes.
Everything tested with the actual Android app from Google play store.

t-echo 2.4.2.5b45303
tbeam 2.5.6.d55c08d
tbeam 2.5.5.e182ae7
tbeam 2.5.4.8d288d5

ConcIusion: I don't think that there is a firmware-problem

tavdog, 2024-12-13:
https://github.com/meshtastic/Meshtastic-Android/pull/1463