meshtastic/Meshtastic-Android #1353: [Feature Request]: Add hardware GPS status to node info card

b8b8, 2024-10-26:
### Tell us your idea.

Suggestion to add the following to the newly implemented node info cards (maybe under the hardware line):
- Hardware GPS model detected at boot
- GPS enabled/disabled status
- GPS satellite lock status
- GPS search status
- Constellations used
This will help with debugging and for nodes without a screen, help the user see if there are any problems with the GPS

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

garthvh, 2024-10-26:
Number of sats is the only value we have in the protobufs.

b8b8, 2024-10-26:
ok, is there a way to turn the satellite icon red when the position interval or smart position tries to send or update position but the GPS does not have a valid position and/or is not detected properly as a half measure?

garthvh, 2024-10-26:
Other than the dop values which has been difficult to interpret the apps know nothing about the gps or position quality.

b8b8, 2024-10-26:
If debug logs are available over bluetooth, could the info be pulled from there?

andrekir, 2024-10-27:
The node details display info that’s sent over the mesh in a `MeshPacket`, like Telemetry data. detailed GPS information is not available.

closed 2024-10-27 by andrekir