meshtastic/Meshtastic-Android #1357: Test button (ext. Notification)

F13crack, 2024-10-27:
### Contact Details

F13-1nfo@outlook.com

### Tell us your idea.

I'd like the app to have a button in the external notification settings, so you can test your settings without needing another device to send you a message.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

cracky22, 2024-10-27:
nice idea. is it possible to have a seperate button which triggers a single "beep" to find the node -> "findMy function..."

andrekir, 2024-11-04:
this would have to be implemented in the firmware. there is no way clients can trigger external notifications coming from the radio.

closed 2024-11-04 by andrekir

cracky22, 2024-11-04:
> this would have to be implemented in the firmware. there is no way clients can trigger external notifications coming from the radio.

that means what do we do now? Should we put this in the issues in the firmware first or do you do that?