meshtastic/Meshtastic-Android #1361: [Bug]: Very Long Slow not an option in Lora settings

jesseas, 2024-10-28:
### What happened?

I tried to update my Heltic V3 to use Very Long Slow and it's not in the drop down under Lora settings (the other profiles are though). I was seemingly able to configure Very Long Slow through the web UI though. 

### App Version

2.5.3

### Phone

Samsung S22 ultra/ Android 14 Oct 1 2024 patch level

### Device

Heltec 3

### Firmware

2.5.4.8d288d5

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

garthvh, 2024-10-28:
Very Long Slow was Deprecated in 2.5

closed 2024-10-28 by garthvh

jesseas, 2024-10-28:
ok, thanks for the info, guess I missed that.