meshtastic/Meshtastic-Android #1364: [Feature Request]: Time limit for metrics graphs

GMart, 2024-10-30:
### Contact Details

Discord

### Tell us your idea.

The metric graphs are great, but depending on how many nodes you have, the time span for the graphs can get very long, making it hard to view the past week of data, for example. I think at least one of these things would be beneficial: 

- Provide a configurable time limit for the metric data. If would delete after this time. This option or button could be entirely within the metric screen if not needed in the actual node firmware. 
- Provide a shortcut to delete data points before a certain date. Mostly the same as option 1.
- Allow the metric screens to be pannable and zoomable with gestures.

Best regards,
GMart

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

b8b8, 2024-10-30:
I think a rolling one week window would be good with a vertical line separating midnight on each day.

GMart, 2024-10-30:
B8, that might be good for a phone, but I think a tablet running Android might be able to comfortably view 2 weeks at a time. That is why I suggested having it configured, or having a delete old metrics button, so it will fill up to the 10,000 log entries or so, unless you press the delete old metrics button.

b8b8, 2024-10-31:
That makes sense. Maybe 4 days in portrait mode and 2 weeks in landscape regardless of device.

closed 2024-11-11 by andrekir