meshtastic/Meshtastic-Android #1394: [Bug]: Switching from a bluetooth device directly to a wifi one does not work correctly.

ragnarblackmane, 2024-11-10:
### Contact Details

dax.marola@gmail.com

### What happened?

I have multiple devices, all connected through BLE, only my heltec is connected through wifi. When I switch from a BLE one to the WiFi one, it looks like it's connected, but can't do any management or send messages, I have to do the following:
1. connected to BLE device;
2. select the None device;
3. then select the WiFi one.

Very rarely this does not work too, then I'll just select None and then again the WiFi device and it works.

### App Version

2.5.7

### Phone

Samsung A35 5G

### Device

Heltec v3

### Firmware

2.5.11.8e2a3e5 beta

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

andrekir, 2024-11-10:
I can't reproduce this. Switching to TCP/IP interfaces works fine on the first try, and reconnects are scheduled correctly if one fails.

TCP/IP disconnects can take up to 90 seconds to detect, check the serial logs on the Heltec for details.

Most common cause for this is trying to connect multiple clients to a single device at the same time. Only one connection is supported at a time.

closed 2024-11-10 by andrekir

tgies, 2024-12-03:
I came here to report this same issue. My wifi node is a T-Beam. When connecting to the wifi node, it appears to download all info from the node successfully, but interacting with the node does not work (sending messages hangs on queued for sending, viewing config hangs at 0% progress, etc) until either some time has passed, or I force stop and reopen the app.

ragnarblackmane, 2024-12-03:
> I came here to report this same issue. My wifi node is a T-Beam. When connecting to the wifi node, it appears to download all info from the node successfully, but interacting with the node does not [...]

Exactly. I don't have any power saving setting set up, the issue is with an Heltec v3.1 and 2 Heltec v3.2.
One of the v3.2 has been updated to the latest beta (2.5.11) from a the previous beta version, the other devices were set up straight to 2.5.11 from the default heltec (non meshtastic) firmware.

From other test its looks like going | BLE on my RAK | -> | None | -> | WiFi on Heltec |  is not always the solution and I have to add another -> | None | -> | WiFi on Heltec |.

Same issue even when going from one Heltec in Wifi to another.
I'll check the logs as soon as I can.
Read somewhere that it may be an issue born from having the nodes on WiFi 2.4GHz and the smartphone on WiFi 5GHz.

sebastianha, 2025-06-29:
Hi,

I would like to reopen this issue because I have exactly the same problem. When connecting to my Heltec V3 via Wifi I cannot send messages after the first connect. I always have to select "None" and then reconnect to the IP to get a working connection.

Any else still having this problem? Should I create a new bug ticket?