meshtastic/Meshtastic-Android #1400: [Feature Request]: increase GPS poll speed

geekscrapy, 2024-11-12:
### Tell us your idea.

Add the option to increase GPS polling speed to at least 1 second. Use case is to track moving vehicles

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

andrekir, 2024-11-12:
There needs to be a balance between polling GPS data and position broadcasts. There's not much point having data you can't broadcast, and every 1 second is way too much for LoRa.

All GPS configs are available in Radio Configuration > Position. Anything lower than 10 seconds probably won't work, but nothing in the app prevents you from using any value.

closed 2024-11-12 by andrekir

geekscrapy, 2024-11-13:
Many thanks.
When you say under 10 seconds won't work, I presume that is due to the limited speed of LoRa?

Taking a look at GPX files produced by other GPS trackers, it looks like logging every 4 seconds is generally accepted, but this is still way under the 10 secs you recommend 😕

geekscrapy, 2024-11-13:
I thought about what I'm actually thinking about doing in more detail.
I'm not actually needing to broadcast onto the LoRa network, it's just to feed into something locally like ATAKS.

So the scenario would be setup the device, connect to the meshtastic app, allow GPS to be received and logged via the local ATAKS on the same phone at roughly 4 seconds. GPS data doesn't actually need to be sent over LoRa, only when ATAKS wants to share it.

andrekir, 2024-11-13:
Yeah, ATAK uses Android GPS directly, not from Meshtastic.