meshtastic/Meshtastic-Android #1407: [Bug]: Android 15 looses Bluetooth connection completely when app in background

Franselbaer, 2024-11-15:
### What happened?

Since Android 15 update on my Pixel 7 recently, the app looses bluetooth connection to T Echo while in background completely and it does not come back at all. Only if I re-open the app, click to "none connected" and then back to my device the connection comes back.

Background authorization is still set (double checked). Battery usage is 2-5% over the day. Seems to be something new in Android 15 and bluetooth...

### App Version

2.5.7 from Google appstore.

### Phone

Pixel 7a

### Device

T Echo

### Firmware

2.5.11

### Relevant log output

```shell
none
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

andrekir, 2024-11-18:
Some users are having Bluetooth issues with Android 15, while most work fine and I can't reproduce it myself. There's not much we can do from the app side to fix that.

closed 2024-11-18 by andrekir

b8b8, 2024-11-19:
its def android and not the app. Im having problems with one of garmin watches as well. Also a pixel 7a.

Franselbaer, 2024-11-19:
The thing is, the app tries to reconnect every 15 minutes if in background
and lost connection to pull everything new..Even that doesn't work well
anymore.

tavdog, 2024-12-13:
I believe I had found a fix in https://github.com/meshtastic/Meshtastic-Android/pull/1463