meshtastic/Meshtastic-Android #1408: [Bug]: Intermittent crashes when clicking Traceroute

jeremiah-k, 2024-11-15:
### Contact Details

@Jack!

### What happened?

We're trying nodes in new areas so I've been running a lot of traceroutes lately.  It doesn't happen every time, but since upgrading to 2.5.9 this morning it's now crashed 3 times.

I haven't had issues at all with previous releases.

### App Version

2.5.9 F-Droid

### Phone

Pixel 6 Pro - GrapheneOS (Android 15)

### Device

T1000-E

### Firmware

2.5.12

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

madeofstown, 2024-11-15:
I'm experiencing this now also. Installed and tested this morning on my mobile node, seemed to be working fine. Now I'm back home and testing it with my G2 and it's crashing every time I run a traceroute.

SimbimChimbetov, 2024-11-16:
Had it few times on RAK4631 but not 100% every trace route.
Also crashes on requesting position on Nodes with fixed Position that are manually set via remote admin.

Android 2.5.9
Firmware 2.5.11

madeofstown, 2024-11-17:
I've now had it crash a couple times while requesting position as well.  Seems that if I force close the app and reopen it will work ok for a bit. Once the app starts crashing on traceroutes it will usually continue until I force close and reopen.

madeofstown, 2024-11-19:
Might have something to do with the node menu refactor? Crashing when I request node info for an unknown node as well

commit c70b0d51 (2024-12-02): fix: remove `SelectionContainer` from `NodeItem`

Removes `SelectionContainer` from `LazyColumn` items causing a crash when calculating positions between nodes that are no longer part of the same layout hierarchy, resulting in:

Fatal Exception: java.lang.IllegalArgumentException: layouts are not part of the same hierarchy
       at androidx.compose.ui.node.NodeCoordinator.findCommonAncestor$ui_release(NodeCoordinator.kt:1210)
...

closes #1408
closed 2024-11-19 by andrekir