meshtastic/Meshtastic-Android #1410: [Feature Request]: Bookmark IPs

rayslinky, 2024-11-16:
### Contact Details

rayslinky@gmail.com

### Tell us your idea.

It would be nice to bookmark IPs that I regularly connect to in the app, instead of having to type it every single time.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

GUVWAF, 2024-11-16:
Are you using ESP32 or a Pico W? 

It should discover IPs of a Meshtastic device automatically via mDNS, but for the Pico W that will only be in the upcoming firmware release and latest Android app.

rayslinky, 2024-11-16:
Neither the Heltec v3 (esp32) nor the T-Beam showed up via IP, just Bluetooth. My use case is also for when I'm away from my network and tunneling in, where mDNS will not work.

GUVWAF, 2024-11-16:
I see, then indeed I think it's a fair request.

Kealper, 2024-11-20:
FWIW: I have my base station node (Heltec V3) connected to my WiFi and it has never been auto-discovered in the Android app on either Android 14 and 15 on a Pixel 8 Pro. I've been using the signed-for-F-Droid builds but I've been downloading the APKs as they get built from here. I actually only found out about the existence auto-discovery stuff when I seen the commit recently that will enable it on the Pico W, I hadn't realized it was supposed to be implemented already in the Android app.

AU518987077, 2024-12-22:
I too would like to have the ability to bookmark my frequently accessed WIFI nodes over their Static IP and my domain and port assignment
would you be able to add both text boxes, and have the connection try a lan connection before it tries a domain connection if both boxes are filled out?

kquinsland, 2024-12-31:
I am a +1 for this (but with a modification): the ability to associate bookmarks with network connectivity.
Some menu or other functionality that allows me to say "when `$homeWifi`, choose `$homeNodeHostname`" and "when `$officeWiFi` choose `$officeNodeHostname`"

Please let me know if that should be it's own issue and I'll be more than happy to file/link!

closed 2025-05-24 by github-actions[bot]