meshtastic/Meshtastic-Android #1413: [Feature Request]: Encrypt Configuration Exports with the private key of the device

Dokument, 2024-11-16:
### Contact Details

adammelton@gmail.com

### Tell us your idea.

I'd like the app to encrypt the configuration when I export it to my local device. Specifically, I would like for it to encrypt the config with the public key of the node I am exporting. 

PKI is good but today it is very difficult to utilize the systems that Meshtastic has in place to verify another user and to maintain security of your keys. Currently configuration exports contain a user/node's public and private key in plaintext which is dangerous*. 

* = Dangerous from a security standpoint. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

b8b8, 2024-11-16:
You can share your configs in the encrypted folder in your phone or for example the ecrypted Note-To-Self in the Signal APP. Password protecting the configs with a long string will make it very tedious to re-import.

Dokument, 2024-11-17:
@b8b8 , Not all phones have an encrypted folder. There would be no reason to have to type the string yourself (on a standard export/import since you would also use the private key of the node to decrypt the configuration file (with an error when this key fails to decode) which is another way to ensure you don't apply the config to the wrong node. Additionally, this feature could be disabled in the UI which would allow for a plain-text export (with warning).

b8b8, 2024-11-17:
If you completely wipe your device you lose that private key and would be unable to then restore the config file without manually copying it back in. The point of the backup file is to restore your keys, needing to keep a list of your public or private key to restore it in the event of a full wipe would be tedious. You also want the ability to to restore to the "wrong node" if say you have one node have admin privileges but has a complete hardware failure (say you sit on your T1000E and break it) then you can have your admin configuration restored to a new T1000E and go on your way. You need to control access to your own files but separately encrypting the config files is not the answer. Its only dangerous if someone else has complete access to your phone and in that case there are far more important things to worry about than Meshtastic configs.

Dokument, 2024-11-17:
> If you completely wipe your device you lose that private key and would be unable to then restore the config file without manually copying it back in.

- Correct. That would be ideal.

> You need to control access to your own files but separately encrypting the config files is not the answer.

- I completely agree, which is why this request is asking for encryption of the config to be a part of the app and not to have to do it separately.

closed 2024-11-18 by andrekir