meshtastic/Meshtastic-Android #1414: [Feature Request]:  Hide the Private Key by default

Dokument, 2024-11-16:
### Contact Details

adammelton@gmail.com

### Tell us your idea.

I would like for the app to hide the node private key by default. Today there is no way to safely share your public key unless you have already copied it from the application in private. Opening the security page reveals your private key in plaintext which would make it trivial for anyone to make a copy by taking a picture or reviewing security camera footage (etc). 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

b8b8, 2024-11-16:
its not like you generally manually type out your public key for someone. the most likely scenario if you every need to give the public key manually is to copy it yourself and send as a text message to someone. Just dont have someone looking over your shoulder when you text it to them

Kealper, 2024-11-20:
I think this could be a good change, to make the private key textbox act like a password box instead of a regular textbox, that way it would have the contents hidden by default but a button on the textbox could be clicked to toggle showing the private key or not.

The web client already does it this way, although I don't know if the iOS client does or not as I don't have any iOS devices to test, so there's at least some precedent for it that I know of.

Dokument, 2024-11-20:
> I don't know that I would ever share my public key by showing someone that page

I am unaware of any other way to access the public key while in the field, excluding via the web interface such as you mentioned. 

An alternate solution could be an option to view the public key for any node from the nodes list, including your own node.  I still want to keep this feature request open since hiding the private key by default really does help security by a lot.

closed 2025-05-24 by github-actions[bot]