meshtastic/Meshtastic-Android #1424: [Feature Request]: Add node public key to "more details" page

Dokument, 2024-11-20:
### Contact Details

adammelton@gmail.com

### Tell us your idea.

I would like to be able to view the public key for other nodes via my mobile app. This is currently not possible today.

Having access to see the public key for other nodes would allow me to verify their key while in the field. Additionally, I could use this field for my own node to share my public key without having to go to the authentication page. Going to the [authentication page reveals my private key along with my public key, both in plain text, which is a security issue](https://github.com/meshtastic/Meshtastic-Android/issues/1414). 

This feature would increase security for senders and receivers, raising the bar for security of the system as a whole.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

b8b8, 2024-11-20:
This is a much better solution. Keep the security menu the same and add the public key to the node card.

andrekir, 2024-12-06:
the Public Key is now shown in the node list key icon dialog: [feat: add Public Key to node key status dialog](https://github.com/meshtastic/Meshtastic-Android/commit/746f155a99a6e34a776c606f8d47f498ac8490d1)

[image]

closed 2024-12-06 by andrekir