meshtastic/Meshtastic-Android #1432: [Enhancement]:  Towards improving UI to parity with iOS and perhaps even better

materoy, 2024-11-26:
### Contact Details

materoy@proton.me

### Tell us your idea.

This is a great app with good stability but the UI is really lacking, especially the home screen with the tabs at the to and settings screen could use some bit of user experience improvement. 
I have started working on improving them and the PR will be available soon. I just wanted to put it here first so that maintainers would know. 

[image]

This screenshot shows my first improvements but more to com. Please let me know if I'm going in the right direction. 

Cheers 
~ Matero

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

andrekir, 2024-11-26:
Thanks for your interest in improving the app! We’re almost done migrating all Fragments to Compose, with only `SettingsFragment` and parts of `ContactsFragment` (ActionMenu) and `MessagesFragment` (ActionMenu and a few components) left.

We'll likely need to finish migrating `MessagesFragment` and `ContactsFragment` before we can replace the ViewPager. How do you plan to integrate your changes with the existing Compose navigation?

materoy, 2024-11-26:
Well then if we are taking the bottom up uproach, I can work on the `SettingsFragment` and migrate it to compose, unless someone else is working on it

andrekir, 2024-11-26:
Sounds good! The "Report Bug" button will be removed, so feel free to take it out during the refactor.

Let me know if you have any questions.

b8b8, 2024-11-27:
My thumbs thank you for wanting to move the tabs to the bottom.

closed 2025-05-15 by jamesarich