meshtastic/Meshtastic-Android #1434: [Bug]: Many translation error in french

HB9HLI, 2024-11-26:
### Contact Details

hb9hli

### What happened?

Hi,  In the latest version, I have many translation errors in french, like mix with other languages. In English it's OK. The previous version don't have this problem. Other friends have same.Many thanks.

Dom. HB9HLI.

### App Version

2.5.10

### Phone

OnePlus 9 pro

### Device

Tbeam/t-echo 

### Firmware

2.5.9.936260f

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

andrekir, 2024-11-26:
https://developer.android.com/guide/topics/resources/multilingual-support#resource-resolution-examples

[image]

There are 196 untranslated words. Missing translations will use resource resolution above.

Contributions are welcome at: https://crowdin.meshtastic.org/android/fr

closed 2024-11-26 by andrekir

Retidurc, 2025-05-05:
Hmm, i'm not sure it's the same issue @HB9HLI was talking about, but a lot of french strings where overriden by another language.

It seems someone on crowdin need to go back to each string one by one to fix the translation 

[image]
Here, for example, all previous variation are good, except from the last one.

Is there any way to mass remove the wrong contributions ?