meshtastic/Meshtastic-Android #1438: [Feature Request]: Possibility to manually apply offset for temperature, humidity and barometric pressure measurements (BME280 and other sensors))

Synthacon, 2024-11-28:
### Tell us your idea.

I'd like the app to...It would be useful to have built-in support (Android app) for applying user-defined offsets to the temperature, humidity, and barometric pressure readings for the BME280 results. This would allow users to easily adjust for systematic inaccuracies or environmental influences (e.g., heat from nearby components).

- Correct for localized heating affecting temperature and humidity readings.
- Adjust for known inaccuracies in factory calibration.
- Enable quick fine-tuning for specific environmental conditions.

Just as example my sensor shows +0.64c and -10% for humidity and currently there is no way to adjust this.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

andrekir, 2024-11-28:
Sensor calibration needs to be handled by the device

closed 2024-11-28 by andrekir

Synthacon, 2024-11-29:
It's not possible to calibrate this sensor on any device. Can only adjust offset programmatically.

Synthacon, 2024-11-29:
> Sensor calibration needs to be handled by the device

It's not possible to calibrate this sensor on any device. Can only adjust offset programmatically.

Synthacon, 2024-11-29:
Posted in firmware development.